FPGAwars/icezum #25: KiCad example loading

gh-jmg, 2016-07-03:
Hi,
Nice project. Checking the nice KiCad ported files you have I see some small issues 

a) SCH files open, but minus some symbols.
    I need to go Preferences.Component Libraries, and add your local library to the list.
   Then, all symbols render OK

b) icezum.KiCad_pcb seems load parts and outline correctly, but looks to be 100% unrouted ? (no traces?)
Or, is it intentional this is unrouted ?

Obijuan, 2016-07-03:
Hi!

Thanks for you feedback

a) Checkout the develop branch and test it again: https://github.com/FPGAwars/icezum/tree/develop

The library should be automatically loaded. If not, please, tell me in this issue

b) The board is currently being routed in the develop branch. All the placement is finished, all the 3D component have modeled (in freecad) and the routing is 20% (only the first layer). I hope that the migration to kicad will be finished in around two weeks

Any feedback is welcome :-)

gh-jmg, 2016-07-03:
Hi Juan,

a) Checkout the develop branch and test it again:
> https://github.com/FPGAwars/icezum/tree/develop
>
> The library should be automatically loaded. If not, please, tell me in
> this issue
>
I'll check this tomorrow.

> b) The board is currently being routed in the develop branch. All the
> placement is finished, all the 3D component have modeled (in freecad) and
> the routing is 20% (only the first layer). I hope that the migration to
> kicad will be finished in around two weeks
>
ok, I was not sure if that unrouted was accidental.Look forward to seeing
the finished design.
If you do this in Altium, you could consider release of both P-CAD and
KiCad formats, as the kiCad PCAD importer is being improved all the time.
That also allows Eagle users to import the design too.

Did you consider doing anything with the Lattice 48 pin QFN iCE5x series ?
They still lack an eval board for those....

Obijuan, 2016-07-03:
> If you do this in Altium, you could consider release of both P-CAD and KiCad formats, as the kiCad PCAD importer is being improved all the time. That also allows Eagle users to import the design too.

Thanks for the tip!

> Did you consider doing anything with the Lattice 48 pin QFN iCE5x series ? They still lack an eval board for those....

I am not sure if that series are supported by the opensource icestorm toolschain. If not, they will be in the future, I am sure

gh-jmg, 2016-07-05:
Hi Juan,

> b) The board is currently being routed in the develop branch. All the
>> placement is finished, all the 3D component have modeled (in freecad) and
>> the routing is 20% (only the first layer). I hope that the migration to
>> kicad will be finished in around two week
>>
> I can see progress in this :)
I'm curious, if you have this routed in Altium, why do you need to re-route
it in KiCad ?
Did the porting/translation pathways have some issues ?

-jg

Obijuan, 2016-07-05:
I am just routing it myself for learning Kicad. Also for learning all the icezum details. The board was routed by Eladio, a college of mine, in Altium. I want to re-do it again in Kicad, for having in mind all the details.

gh-jmg, 2016-07-05:
HI Juan,

I am just routing it myself for learning Kicad.
>
Ah, that makes sense.

> Also for learning all the icezum details.
>
The board was routed by Eladio, a college of mine, in Altium.
>
I want to re-do it again in Kicad, for having in mind all the details.
>
You should be able to import from Altium, but I guess it depends on how
many changes are also merged in.
I've found KiCad seems to tolerate small end alignment deviations, (ie
slightly off exact pad centre)

I'm also new to KiCad, but it is growing on me,
The Shove router is very nice, and has better cleaning algorithms than
Mentors.
The mouse wheel navigation with rate-based zoom is also novel, and the more
I use that, the more adept I get and the more I like it. One hand for
navigate and one hand for commands, is nice.

gh-jmg, 2016-07-22:
Hi Juan,

I see a finished? design recently posted - looks nice :)

Some comments/observations

C31 is labeled 4.7nF, but maybe meant 4.7uF ?

When I use MicroUSB, I choose ones with through-holes (eg 4x)  on the outer
framework.
Usually those are short, (< 1.6mm), but give both location and more
strength and less likely to 'rip-off' from accidents.

regards
Jim G.

​

Obijuan, 2016-07-24:
>  see a finished? design recently posted - looks nice :)

Thanks! It is almost finished

> Some comments/observations C31 is labeled 4.7nF, but maybe meant 4.7uF ?

Yeah! It is a bug. I've already fix it in [this commit](https://github.com/FPGAwars/icezum/commit/6cd7838ead1d836aee266f3c594ddf4b5326ba79).  Thanks a lot for reporting:-) 

> When I use MicroUSB, I choose ones with through-holes (eg 4x) on the outer framework. Usually those are short, (< 1.6mm), but give both location and more strength and less likely to 'rip-off' from accidents

Thanks for the advice. It makes more sense to use your MicroUSB.  We will use them in future versions

I will close this issue. If you have further comments or bugs, please do not hesitate on filling a issue

Thanks for you help  ;-)

closed 2016-07-24 by Obijuan

gh-jmg, 2016-08-16:
Hi Juan,

Using this design as a nice test for kiCad scripts, I notice a couple of
things :

a) There are a few thermal vias, that have imported as parts from Altium

This gives a skew to the BOM and netlist
BOM has ~ 26 extra parts like this...
VIA**   pad-via
VIA**   pad-via
Netlist has 26 extra nodes
VIA**.~

BOM and Netlist will be cleaner, & need less manual cleanup, if those
Thermal Vias can be swallowed into the footprints ?
​
b) Some entities are modules with zero pads, so I added a test to skip
these.
(These can also confuse the BOM )
These may be just dummy place keepers ?

*REMARK* Skipped doc1 (has 0 Pads)
*REMARK* Skipped DOC1 (has 0 Pads)
*REMARK* Skipped doc7 (has 0 Pads)
*REMARK* Skipped doc2 (has 0 Pads)
*REMARK* Skipped doc3 (has 0 Pads)
*REMARK* Skipped doc10 (has 0 Pads)
*REMARK* Skipped DOC3 (has 0 Pads)
*REMARK* Skipped doc4 (has 0 Pads)
*REMARK* Skipped doc5 (has 0 Pads)
*REMARK* Skipped doc6 (has 0 Pads)
*REMARK* Skipped doc8 (has 0 Pads)
*REMARK* Skipped doc9 (has 0 Pads)
*REMARK* Skipped JPM1 (has 0 Pads)
*REMARK* Skipped DOC4 (has 0 Pads)
*REMARK* Skipped DOC5 (has 0 Pads)
*REMARK* Skipped G*** (has 0 Pads)
*REMARK* Skipped REF*** (has 0 Pads)
*REMARK* Skipped DRILL2 (has 0 Pads)
*REMARK* Skipped DRILL4 (has 0 Pads)
*REMARK* Skipped DRILL3 (has 0 Pads)
*REMARK* Skipped DRILL1 (has 0 Pads)

regards
Jim G

Obijuan, 2016-08-16:
Hi @gh-jmg !

I've created two new issues to discuss there: #28 and #29 

Thanks a lot!

gh-jmg, 2016-08-18:
Hi Juan,

FYI, I just came across this
http://www.zofzpcb.com/

A 3D Gerber viewer, quite impressive.
I plotted using Altium GBR names, which it seems to better sense of, & I
drop a .D356 file into the GBR directory.

It can also load IPC356, and check a design.

So,  I feed latest Icezum into this.... just to see what happens...

See attached image - it has mostly inferred correct 3D info :)

The Via** parts have confused it a little, as it thinks they are all *one*
part, so it makes a  single, large rectangle covering all Via**, and it
also thinks they are pins, so renders as that, with 'bonus spines'.

Moving those thermal vias into footprints, (or even just as manual Vias) as
suggested before, should clean that up.

It renders non-trace holes differently from trace-holes for some strange
reason, I think a ZofzPCB bug ?

It renders U6 with a strange angle spin, not sure why ?

Hmm, looking closely, U6 is unusual, and it has guessed a ~SO16N (almost,
pin pitch is *not* same on each side ?), and maybe the pin-centroid is
skewed (tho centroid seems square to me ?)
That may be a ZofzPCB quirk, but is is doing a quite good job of inferring
3D onto the Gerber + IPC info

Attached is the Screen grab, and the IPC error report.

​I see some Gerbers from your github have teardrops - were those Kicad
plotted ?

There is this
https://forum.kicad.info/t/yet-another-python-teardrop-script-adds-and-deletes-teardrops-to-a-pcb-v0-3-0/3388
I suggested he add Curved Teardrops, and they are now in :) - I've not
tested this yet.

ICP356 Netlis v.s. PCB copper test
965 total testpoints processed

27 of No-copper testpoint errors
No copper for Testpoint +5V_P layer both at (41.7,  43.8)mm
No copper for Testpoint +5V_P layer both at (39.1,  43.8)mm
No copper for Testpoint +5V_P layer both at (36.6,  43.8)mm
No copper for Testpoint +5V_P layer both at (34.0,  43.8)mm
No copper for Testpoint +5V_P layer both at (31.5,  43.8)mm
No copper for Testpoint +5V_P layer both at (29.0,  43.8)mm
No copper for Testpoint +5V_P layer both at (47.0,  9.5)mm
No copper for Testpoint +5V_P layer both at (49.5,  9.5)mm
No copper for Testpoint +5V_P layer both at (52.1,  9.5)mm
No copper for Testpoint +5V_P layer both at (54.6,  9.5)mm
No copper for Testpoint +5V_P layer both at (57.2,  9.5)mm
No copper for Testpoint +5V_P layer both at (59.7,  9.5)mm
No copper for Testpoint +5V_P layer both at (63.5,  43.8)mm
No copper for Testpoint +5V_P layer both at (61.0,  43.8)mm
No copper for Testpoint +5V_P layer both at (58.4,  43.8)mm
No copper for Testpoint +5V_P layer both at (55.9,  43.8)mm
No copper for Testpoint +5V_P layer both at (53.3,  43.8)mm
No copper for Testpoint +5V_P layer both at (50.8,  43.8)mm
No copper for Testpoint +5V_P layer both at (48.3,  43.8)mm
No copper for Testpoint +5V_P layer both at (45.7,  43.8)mm
No copper for Testpoint NET-(J5-PAD1) layer both at (27.9,  2.5)mm
No copper for Testpoint +5V_P layer both at (64.2,  28.4)mm
No copper for Testpoint +5V_P layer both at (64.2,  25.9)mm
[... 469 lines cut ...]
    U8-17    J12-6    J12-6    J11-6    J11-6    R22-1
subnet 2:
    J4-10    J4-10    J12-6    J12-6
IPC356 net "POWER" is broken into 3 subnets:
subnet 1:
    TP6-1    J5-8    J5-8
subnet 2:
    Q5-2    C60-1
subnet 3:
    Q2-4    Q2-5    Q2-6    Q2-6    L4-1    C66-1
IPC356 net "SYS_RESET" is broken into 2 subnets:
subnet 1:
    SW4-1    SW4-1    R35-1    R33-1
subnet 2:
    J5-3    J5-3    SW4-1    SW4-1
IPC356 net "USB_5V" is broken into 2 subnets:
subnet 1:
    TP7-1    F1-2
subnet 2:
    Q3-2    R39-2    R38-2
IPC356 net "U_SW1" is broken into 2 subnets:
subnet 1:
    SW1-1    C34-2
subnet 2:
    U1-10