FPGAwars/icezum #26: Migration to Kicad

Obijuan, 2016-07-24:
Migrate icezum V1.0 to Kicad. It will be the V1.0K release

easyw, 2016-07-29:
Hi,
I downloaded your kicad project and I noticed that sw_smd_6.0x3.8.wrl is not displaying in the 3d viewer latest kicad dev release...
I just noticed that was exported directly by FC and not using kicad StepUp...
I tried exporting it with ksu and everything went fine...

thanks for using kicad StepUp tools 👍 
Maurice

Obijuan, 2016-07-29:
Hi easyw! Thanks a lot!  Could you please emit a pull-request with the correct file? that would be great! Thanks a lot :-)

Obijuan, 2016-08-16:
Hi @easyw , your issue has been open as a new one (BUG). The migration to kicad issue will be closed

closed 2016-08-16 by Obijuan