FPGAwars/icezum #27: sw_smd_6.0x3.8.wrl is not displaying in the 3d viewer

Obijuan, 2016-08-16:
Reported by @easyw

Hi,
I downloaded your kicad project and I noticed that sw_smd_6.0x3.8.wrl is not displaying in the 3d viewer latest kicad dev release...
I just noticed that was exported directly by FC and not using kicad StepUp...
I tried exporting it with ksu and everything went fine...

thanks for using kicad StepUp tools :+1:
Maurice

Obijuan, 2016-08-26:
Hi @easyw !

I've just upload a new sw_smd_6.0x3.8.wrl exported form the kicadstepup. It should fix the bug, but I have not tested it yet with the new kicad viewer.

I will close the issue. If you test it and find that the bug remains, please reopen it or let me know

Thanks a lot

closed 2016-08-26 by Obijuan

easyw, 2016-09-08:
it seems fine to me 👍 
thank you
Maurice