FPGAwars/icezum #33: Migrate the v1.1 schematic to kicad

Obijuan, 2016-08-31:


AntonioMR, 2017-10-25:
Hello
I have almost finished migrating the schema from version 1.1 to kicad.
I am working on my local branch v1.1kicad not merged with local master.
is it okay if I send pull-requests on this branch, to merge with remote/master only when complet the migration? any other suggestions?

Obijuan, 2017-10-26:
Great! Yes! That is perfect! :-)  go ahead!!

AntonioMR, 2017-10-26:
Hello again,
It's seems I'm not able to send a Pull Request over a branch that it's not previously created, or I don't know how to do it. A possible workaround is:
- You create in your repository a local branch v1.1-kicad and push it to the icezum repository, so I will be able to send a PR to that branch.
- I push my local branch to an v1.1-kicad branch on my icezum repository fork. You fetch this branch on an local branch in your repository and then you push it to the icezum repository in github.
Do you know any other solution? Please let me know.

Obijuan, 2017-10-26:
I've already created the v1.1-kicad branch from master. Try to commit there

AntonioMR, 2017-10-26:
Thanks,
Done!

Obijuan, 2017-10-27:
I think this issue is fixed and can be closed.  Thanks a lot for your contribution! :-)

closed 2017-10-27 by Obijuan