FPGAwars/icezum #37: order?

phil123456, 2017-02-14:
hi ,

I cant find any web site to order these....

Obijuan, 2017-02-14:
Hi,

Currently is not on sale. We are organizing small local crowdfundings for funding the batches. Batch 1 was finished (110 units). Now we are manufacturing the Batch 2 (250 units). The backers are helping us on the FPGAwars list (but i is only in spanish).

Is it possible that for the following batches we launch a more global crowdfunding compain

moebiussurfing, 2017-03-23:
Hola buenas, me gustaría participar en el crowfounding y adquirir alguna unidad, es posible? gracias

1138-4EB, 2017-03-24:
> Hola buenas, me gustaría participar en el crowfunding y adquirir alguna unidad, es posible? gracias

https://groups.google.com/forum/#!topic/fpga-wars-explorando-el-lado-libre/a5mmxIj4IcM
https://groups.google.com/forum/#!topic/fpga-wars-explorando-el-lado-libre/wOaLSpkVSII

closed 2017-03-27 by Obijuan